alephobjects/Cura #74: Tooltips are not appearing on windows and linux

kakaroto, 2015-01-01:
It looks like commit 8516605251c7e356ed0d6ba4252408fe0440919b caused tooltips to stop working on both Linux and Windows. I tested that commit and the one before and the bug appears in that one.
Since Justin says it fixes an issue with tooltips on Mac, I suggest making those changes only for Mac and leaving the old way of displaying tooltips for non-mac platforms.

Jnesselr, 2015-01-01:
I'll look into it and try to fix the original issue since I'm not overly
fond of how Mac handles the tooltips either, and because I understand the
framework a little better now.

reprapperMC, 2015-01-02:
Building from master I am seeing the tool tips correctly on Debian and Ubuntu. Are you hovering over the field or the field label?

reprapperMC, 2015-01-02:
Tool tips are confirmed to be working on Linux and Windows as expected. They do only appear if you hover over the field label text, but it does work.

closed 2015-01-02 by reprapperMC