alephobjects/Cura #75: Machine settings and about window hiding at first use, OSX

reprapperMC, 2015-01-02:
On build 1.15 and the most recent from AO-build, the machine settings and about windows do not appear as expected. To see either window you have to, after selecting the option in the top menu, click the Cura application in the dock. After the window appears this first time you can close it and open either window as is normally expected.

Jnesselr, 2015-01-07:
This one has given me quite a bit of trouble, mostly because debugging the issue causes it to disappear, and the original issue doesn't occur on my computer.

The issue is that the about window, machine settings, and preferences dialog all don't pop up on some systems. If you clicked the Cura icon in the dock, it would show the window you wanted correctly. I think it has something to do with the parents of the window being set to nothing.

If you run the packaged in the terminal, it works just fine. If you run it from the disk image, no problem. If you run it from the Applications directory, the window fails to show up until you click on the icon. Clicking on the icon sends the wx.EVT_ACTIVATE_APP, but that doesn't seem to help the issue. Whether the handler is there or not doesn't seem to matter at all.

Jnesselr, 2015-01-07:
Weird workaround here: http://stackoverflow.com/a/10922735/2039051

The reason was that the mac application isn't 'active' when it first starts up. There's a non-hack workaround somewhere, I just have to find it. Technically, the code that this replaced should have been the solution to our problem, but it wasn't working.

closed 2015-06-17 by reprapperMC