alephobjects/Cura #78: Shaders and VDO not properly released

kakaroto, 2015-01-06:
Not sure what this is about but whenever I close cura, I get these printed to stdout :

    Closing down
    Shader was not properly released!
    Shader was not properly released!
    Shader was not properly released!
    VBO was not properly released!

Jnesselr, 2015-01-07:
This is definitely related to OpenGL: https://github.com/alephobjects/Cura/blob/AO-build/Cura/gui/util/openglHelpers.py lines 95 and 236.

This commit may have something to do with it, but I doubt it: https://github.com/daid/Cura/commit/c5109fb2e477e71acf611c05c30e389813867f63

Basically, somewhere is not calling the 'release' function on a GLVBO object and a GLShader. In Cura/gui/sceneView.py, the release function is called only once on a GLShader object, and that's when it needs to get rid of the old one so a new one can be there.

When the sceneView is removed, it needs to clear out _objectLoadShader, _objectShader, and _objectOverhangShader. Those are the shaders that aren't being released properly. I'm adding the function to release them on shutdown in a bit.

As for the VBO, I don't always get that error, so I don't know what it relates to. Maybe it's related to a specific mode that you're in when you quit?

Jnesselr, 2015-01-07:
These can be merged together when merging with the AO-build branch. I didn't expect to be able to solve the VBO issue when I did the first commit, but I had already pushed it upstream when I figured the VBO issue out.

kakaroto, 2015-01-08:
Looks like this is fixed, thanks Justin

commit 8a78e784 (2015-01-07): Fixes #78: VBO properly released
closed 2015-01-08 by kakaroto