alephobjects/Cura #79: Pronterface UI window wasn't able to be closed on OSX ML

orias, 2015-01-16:
A recent Mini customer was not able to close the PronterfaceUI window. 
Steps:

* Open Cura, (simple mode)
* Load model
* Press Control (window opens and has focus)
* Change from Simple mode to Full settings
* Clicked on the various tabs
* Press Expert menu
* Muck about with the expert settings
* Unable to close PronterfaceUI
* Close Expert settings
* Unable to close PronterfaceUI
* Close Cura
* PronterfaceUI closes.

orias, 2015-01-16:
14.09 1.18

kakaroto, 2015-01-28:
I've tried this on Linux and wasn't able to reproduce the issue.

reprapperMC, 2015-06-17:
I'll test this in the new 15.02 build

Jnesselr, 2015-06-17:
This was related to sleep settings on the Mac.

This also means that if a function throws an exception while the window is
closing, the window will not close. This should be a separate issue in
itself.