alephobjects/Cura #81: Cura

reprapperMC, 2015-01-29:
From a user:
"This has happened to me twice now, so I believe it is a bug? The printer suddenly stopped during a large print and I couldn't get it to resume. Screenshot of the error attached. Any input would be appreciated! I don't think it's me accidentally unplugging anything."

It appears they were using OSX but have not gotten confirmation of the version. They did see a checksum error in the print control UI terminal, "Error:No Line Number with checksum, Last Line: 56"

They mentioned this post: http://umforum.ultimaker.com/index.php?/topic/2371-term-tab/

jebba, 2015-01-29:
What did the screenshot have? If it had lines like in the ultimaker post, ala:

Error:Line Number is not Last Line Number+1, Last Line: 65110
Error:checksum mismatch, Last Line: 66064

That likely is a different issue than the print hanging. Hanging is more likely to be caused by USB, Mac going to sleep, etc.

Jnesselr, 2015-01-29:
And this is happening with Cura Lulzbot edition? I thought we fixed the Serial comm issues, but it seems that it might have something to do with the hardware, as someone had the same issues off of the SD card.

I agree that it could be related to sleep issues, but the only one that currently doesn't support sleep is Linux.

reprapperMC, 2015-01-29:
I just put the information they gave. It is possible there is something up with the STL that Cura Engine is failing on. The print failure occurred both time on the same STL.
[image]

Jnesselr, 2015-01-29:
I'm noticing with these errors, it's always around line 56. I've gotten ±3
on that, but still. We could be losing the secondary process that it
creates. It's a possibility that the process actually doing the serial is
allowed to sleep when the primary process is the one that isn't sleeping.

I'd find out if saving to gcode then running with something else would
cause the issue.

jebba, 2015-01-29:
Did it fail in the exact same place?

reprapperMC, 2015-01-29:
We have the question out to the user to see if the failure occurred in the same place on both prints. The STL that the failure occurred on is http://www.thingiverse.com/download:581548.

reprapperMC, 2015-01-30:
Below is the gcode used in the failure; however, I do not think that will give any thing new. With further feedback, the failures all appear to be around the same time in the print. They have said they have only seen the failure in prints of three hours or more and have seen it with other models. I am still waiting on confirmation that it is OSX and the version of OSX. I will do some further testing tonight, but I have previously been able to print long extended prints on the OSX build.

http://devel.lulzbot.com/tmp/T-RexSkull_Jaw_SimplePolygons__repaired_.gcode

Jnesselr, 2015-01-30:
I've done a couple of four hour prints on it with the mac with no issues.

reprapperMC, 2015-02-02:
I was able to successfully print the T-rex skull jaw using the HIPS, Normal Quality print setting in the simple mode (3:20 hrs) on the OSX build of 12.09-1.18. This was on OSX 10.10 with all of the energy saving settings turned on. However, even though the print completed all the way through, Cura was in a non-responsive state. I did find two logs in the console log:

2/1/15 8:03:33.820 PM ReportCrash[71004]: Invoking spindump for pid=51017 thread=637463 percent_cpu=50 duration=180 because of excessive cpu utilization
2/1/15 8:03:34.511 PM spindump[12563]: Saved cpu_resource.diag report for Cura version 14.09 (14.09) to /Library/Logs/DiagnosticReports/Cura_2015-02-01-200334_ao-osees-MacBook-Pro.cpu_resource.diag

Tomorrow I will run the test again with a fresh Cura install to see if it occurs again.

jebba, 2015-02-02:
I saw a few long (8+ hour?) prints over the weekend on OSX that worked fine. I did a couple 10+ hour ones on Debian and they were fine too, fwiw.

jebba, 2015-02-08:
I've done many 4+ hour prints and have seen long ones on a Mac too. I just finished a 31 hour print on Debian without problems. I haven't seen this issue, fwiw.

kakaroto, 2015-02-10:
It could be something unrelated to Cura, OSX and power saving. To me it sounds more like someone using PLA and having heat creep cause a jam. I know that the gcode shows temperatures of 240 which rule out PLA and the Mini has active cooling fan on the extruder, which rules out heat creep, but still, that's what it sounds like, so it's still possible the error here is unrelated to the actual software.

As for the checksum error, as discussed on IRC with reprapperMC, that's unrelated, every print (which isn't using the fix in master) will have that outputted right at the beginning, and the line 56 +- 10 is because that's how long it takes for the Marlin buffer to fill up and show the error.

jebba, 2015-02-12:
> To me it sounds more like someone using PLA and having heat creep cause a jam

It doesn't sound like this *AT ALL* to me. Plus there is no heat creep on the Mini.

I have seen the printer keep running, but the screen doesn't refresh after a long time (just the temp chart refreshes). I haven't noticed this on Mac, but I'll see if it comes up.

kakaroto, 2015-02-12:
> It doesn't sound like this AT ALL to me. Plus there is no heat creep on the Mini.

Yeah, my bad, a heat creep issue would keep printing but without extruding, in this user's case, it just stops printing, which is a different behaviour. Anyways, I knew that wasn't it since I said it wasn't PLA and the Mini doesn't have that issue, but was just wondering if the issue could be related to some other factor and not just Cura itself.

Cura being unresponsive would cause this issue though. Any idea how to reproduce it ?

jebba, 2015-02-13:
I just finished a 2 day print and the Cura interface won't refresh, except for the temp window. The print turned out awesome though. This on Debian.

Jnesselr, 2015-05-13:
I'm not 100% on this because I literally have no logs indicating an error at all, but my TAZ just literally died in a print 7 hours in to a 10 hour print. While this is highly upsetting, the weirdest part is that it wasn't running in Cura, but through BotQueue. What's even weirder is that it caused both BotQueue to crash.

It could be that the device itself is disconnecting and it might not be an issue with the client.

jebba, 2015-05-13:
By the way, the original poster used another Mac and had no problem with it (actually totally loved it). If you think there is a particular bug that caused your failure, start a new ticket for it. I think this one is done. Thx.

closed 2015-05-13 by jebba