alephobjects/Cura #84: Cura becomes unresponsive during prints on OSX

reprapperMC, 2015-02-09:
During longer prints OSX, Cura becomes unresponsive. The print will finish and complete but after the print you will find Cura is in an unresponsive state and has to be force quit.

I need to do further testing as the only testing I've done to reproduce this is with all of the energy saving settings turned on.

Jnesselr, 2015-02-09:
I don't think it has to do with energy savings. I have had it happen before
that was implemented. It seems like it is because of how the serial
communication is handled. The second process hangs, then the first one
hangs too. I've seen the second process hang without the first one hanging
though. When that happened, I tried putting the second process into a
thread and causing the first to yield. This seemed to alleviate issues. We
may need to do something similar for the main control window.

jebba, 2015-02-25:
This is (newly) reported happening on a Mac running 10.9.3. The print turns out fine, but at the end, cura needs to be force quit.

nickthetait, 2015-04-27:
I have consistently experienced this issue with Lulzbot Cura 14.09-1.18 on Ubuntu for prints over ~3 hours long. This is specific to the control/print window. You can still open/modify models and start new prints from the planning window.

nickthetait, 2015-04-27:
Believe that this issue is present upstream: https://github.com/daid/Cura/issues/1115

UASspecialist, 2015-06-22:
Hello I am running yosemite 10.10.3 (14D136) IOS and are having a few issues one being Cura becomes unresponsive during prints that are long duration "overnight" is there a fix for this? One other thing was that the drop down menu for print support and brim are sometimes I.e they don't drop down and require more than one click sometimes, CURA is what I am using My printer is the Taz 5 LuLZE bot. Anyone know of a fix, a better UI for using with my printer? 

Thanks Clancey

Jnesselr, 2015-06-22:
While we could suggest other UIs to use, an issue tracker probably isn't the best place. I'm still not 100% certain why this error is triggered, but I think it is how the serial comm is split into a separate process than the main application itself.

I'll check up on this with the upstream issue.

nallath, 2015-06-23:
Like I said in the upstream, We've basicly redone -everything- inside Cura, which can be found here; https://github.com/Ultimaker/cura. It should be much easier to maintain this version as it supports plugins, which should make it a lot easier for other people to contribute. 
I've cleaned up the serial printing for that version of cura, but it's not perfect, as we (read; Ultimaker) are moving away from this. But we would accept any fixes that you guys make.