alephobjects/Cura #85: Loading gcode for mini outputs unknwon command errors

kakaroto, 2015-02-10:
When generating a gcode file for the Mini and then loading the gcode file into Cura, we get a few "Unknown G code" or "Unknown M code" errors printed on the terminal :

    Unknown M code:204
    Unknown G code:29
    Unknown M code:204
    Unknown M code:400

I'm not sure how the input file gets parsed but I even ended up having my terminal spammed by thousands of "Unknown M code: 204" lines when I left it overnight (note: only the 204 was repeated, not the 29/400 commands).
Adding support for those commands (just ignoring their presence maybe?) is required since those commands appear in the start gcode for the mini profile.

Jnesselr, 2015-02-10:
I don't really understand what M204 does for us, but G29 is the auto bed
leveling routine and M400 is a command to wait for all moves to finish
essentially clearing the buffer. I originally thought this might be coming
from the firmware, but the firmware should support the G29 command. If it
isn't the firmware, why is cura restricting gcode at all? Shouldn't it just
send it? I'm assuming it's sending it anyway, just complaining about it.

kakaroto, 2015-02-10:
Yes, it's sending it, but it's not complaining about it when sending gcode to the printer, it's complaing about it when it's parsing the gcode in order to build the 'layers' view, just load a gcode file, or an stl then switch to layers view to see the errors, otherwise, it doesn't complain.

kakaroto, 2015-02-10:
Should be fixed now.

closed 2015-02-10 by kakaroto