alephobjects/Cura #89: Users upgrading to 1.18.7 with TAZ 4 machines will lose their quickprint settings

kakaroto, 2015-02-12:
IRC quote :

```
<KaKaRoTo> reprapperMC, I see an issue with the TAZ 5 stuff
<KaKaRoTo> you changed the name of the machine
<KaKaRoTo> so anyone using a previous version of Cura and suddenly upgrades, will have his quickprint settings fall in the 'other' category
<KaKaRoTo> because their profile.machine_type will not have been updated
<KaKaRoTo> so they'll either be forced to delete the machine and add it again so it gets set to lulzbot_TAZ_4 or they will be generating with it gcode with 'default' settings and wondering why the prints aren't as good as before
<KaKaRoTo> my advice would be to keep the machine_type profile name for TAZ4 as "lulzbot_TAZ" and just add the TAZ 5 as "lulzbot_TAZ_5"
```

Basically, the machine_type profile name needs to be reverted back to the previous value of "lulzbot_TAZ" otherwise users upgrading their Cura installation will not be able  to use TAZ quickprint settings.

jebba, 2015-02-12:
Ya, sounds like lulzbot_taz will be for 1-4 and do a new one for 5.

kakaroto, 2015-02-16:
Fixed in master and lulzbot-stable branch. Closing.

closed 2015-02-16 by kakaroto