alephobjects/Cura #97: Debian Jessie wxpython version

nickthetait, 2015-04-07:
Between Wheezy and Jessie, Debian upgraded a wxpython library from 2.8 to 3.0 introducing some problems. Cura compiles/runs with the new version, but every time you click a GUI element a duplicate window is opened so you cannot accomplish anything. You also get PyDeadObjectErrors in the terminal. This only impacts installations starting at Jessie.

http://docs.wxwidgets.org/3.0/overview_changes_since28.html

While the old version is deprecated it does not seem worth making changes to Cura since: ["You can co-install wx 2.8 and 3.0, so applications can switch independently of one another."](https://wiki.debian.org/Teams/WxWidgets/Transition2.8to3.0) Also we plan to swap off wxwidgets onto something newer like Kivy.

This is not necessarily a reasonable course of action, just my idea. Thoughts?

jebba, 2015-04-07:
The (admittedly lame) reason we are seeing this late in the game is our Jessie systems were nearly all upgrades from wheezy, and not fresh jessie installs. So the wxgk2.8 and related packages were still on the system. They appear to be working, but maybe there are hidden issues we haven't noticed. That's doubtful due to how extensively we've used it with Jessie. It doesn't cause any dependency (apt) problems from what I've seen. We could check with CSR to see if they know of any issues. One option would be to simply copy the necessary wxgtk2.8 .debs and source from wheezy and plonk them into our private aodeb repo, where we have cura. Then people would do an apt install of cura, and it would drag in copies of 2.8 from our apt repositories:

http://download.alephobjects.com/ao/aodeb/

http://devel.alephobjects.com/ao/aodeb/

orias, 2015-04-07:
We haven't seen any Debian Cura issues yet in the field.

nickthetait, 2015-04-27:
Looks like Daid fixed this about two weeks ago. https://github.com/daid/Cura/commit/619a0504c508d04440b47881bd72573009672a3b

closed 2015-06-04 by nickthetait