alephobjects/Cura #98: Print center of model when scaled to max print area is not in bed center

orias, 2015-04-08:
[image]

The model is scaled to max print area then gcode is saved.

The machine profile on TAZ5 has the correct travel dimensions of: 
X 298 Y 275 Z250

The print is offset on the X axis by 1cm to the right, in the direction of X max.

Karrad, 2015-04-08:
I was able to fudge a fix.

Scaled model to max using standard machine settings. Was able to then reduce X travel length by 20mm, w/o resizing the object. This placed the bed leveling print in the center of the bed.

nickthetait, 2015-05-05:
Is this just for the Taz? Created two test objects maximum size for the mini/taz, changed settings to no infill/top/bottom, then printed. I get a perfect square on the mini. But for the taz the print is shifted to the right (X+) 5mm, such that the right edge is nearly falling off the bed. Centering along Y axis looks prefect.

kakaroto, 2015-05-05:
Either the endstop on your TAZ is not properly calibrated, or the dimensions are wrong (by 5mm ?) in the taz settings in Cura.
It really depends I guess on whether or not the distance between the square and the border of the print bed is the same in the X min and the Y sides, or if it's not.

nickthetait, 2015-06-24:
I've run into other users that have complained about this. Definitely specific to the Taz, dunno if it is Firmware related at all.

kakaroto, 2015-06-26:
I can confirm the issue is caused by the machine settings along with the endstop position.  Basically, the print bed is 300x300 mm, and the real usable area is the print bed area minus the wasted space due to the endstops. For the Y direction, the Y endstop puts the nozzle at roughly 12.5 mm from the edge of the bed, that means the printable area is 300-12.5 = 287.5 mm, from the home to the edge of the bed.. if we leave the same 12.5 mm on the other side (to center the print), that gives us 300-25=275 mm print bed depth. Which is exactly what the machine settings shows.
The nozzle is however about 5 mm from the edge of the bed in the X direction. So if we want to center the print, we need to have the print width set to 300 - 5*2 = 290 mm. However, it's currently set to 298 (which means the endstop was planned to be 1 mm from the edge, but it's not). 
I've set the print bed width to 290 mm in the machine settings, and printed a max-scaled cube and the print was perfectly centered in my print bed. as it was 13.2 mm from either edge.

Basically, the solution would be to change the bed size to 290 on all taz machines.

closed 2015-06-30 by nickthetait