alephobjects/Cura #102: Pause at Z plugin

Karrad, 2015-04-23:
If the pause at Z plugin is enabled, it will cause the Mini to park the head immediately after the wiping process. This plugin will remain activated, even after switching back to quick print mode.

Video: https://drive.google.com/file/d/0B8CqWMpVrmm4c2h0YmdhaEh1OVU/view?usp=sharing

As we cannot resume the print after pausing, it may be best just to remove this as a standard plugin to prevent issues.

nickthetait, 2015-04-23:
I can confirm that this is broken. Discovered it yesterday!

On my Mini I was able to print until the first pause. At which point there is no way to resume, so I clicked "print" which started the print over again.

Not sure if this affects TAZ.

Karrad, 2015-04-23:
It does effect the TAZ as well. I believe all Ultimaker printers have a resume button directly on the printer, which allows them to resume after pauses.

All attempts at manually entering the resume Gcode command has been unsuccessful. The closest we have come to being able to pause and resume is the G4 SXXX command. (dwell)

Jnesselr, 2015-04-23:
Pause at Z uses M0, which requires either a timeout or button press. If we use M1, then the next gcode command sent would wake it up.

reprapperMC, 2015-04-24:
Rather than removing the plugin completely, it may be better to make the plugin not available when the LulzBot Mini is selected as the active machine.

kakaroto, 2015-04-28:
I'd have to disagree here on the issue.
The first issue reported by Karrad seems to be very different. You can see the code from the pauseAtZ plugin that jnesselr linked to, and you'll notice that it doesn't do anything until it finds at least the first ";LAYER" line in the gcode, and in the video of Karrad, the Mini stopped before the auto-bed-leveling sequence, which means it stopped at the start.gcode, which should be free of any interference from pauseAtZ.

I have also checked the Marlin code, and both M0 and M1 are considering the exact same commands, there is no difference in how they are both handled (regardless of what it says on the wiki), both will wait until the LCD is clicked, and both will accept the S and P keywords for a timeout. Also, both commands are disabled if the firmware was not built with LCD support, which means that the M0 and M1 commands are simply ignored on the Mini because it doesn't have an LCD. I have just tested this, and it does work, the print head gets parked and right away printing is resumed.
As for the TAZ, I've used M0 before and clicking on the LCD's encoder will resume printing, so I think that the reason Karrad thinks different is caused by something else as stated before.

The one issue I've found witht he pauseAtZ on the Mini is that the default park position is 190x190 while the mini has 155x155 area, which means it tries to park beyond the limits of the bed, so when it resumes, the next layer is shifted compared to the previous layers.

kakaroto, 2015-05-22:
I've added a Pause/Resume button to the UI when the print is running, and made the M0/M1 commands be considered as a Pause, and the PauseAtZ plugin works just as expected now.
We only need to change its different park position to something closer to 140 rather than 190, otherwise on the mini, the print will be shifted.

closed 2015-05-22 by kakaroto