alephobjects/Cura #104: Print one object at a time does not work

nickthetait, 2015-04-29:
Steps to reproduce:
1. Start Cura
2. Add any LulzBot machine
3. Load more than one model
4. Click "tools" -> "Print one at a time"
5. Swap to layers view
6. Slide through all the layers

The problem: All objects are sliced together "Print All At Once" with visible transitions (in blue) between models.

nickthetait, 2015-04-29:
I think the problem is that none of the LulzBot printers include printer head size settings. Since Cura has no safe way to plan head movements it reverts to the safest method.
https://github.com/daid/Cura/issues/1081

These settings can be added [here](https://github.com/alephobjects/Cura/blob/LulzBot-stable/Cura/gui/configWizard.py#L447), but I do not know what the correct numbers are. I took a few quick measurements on my Mini:
extruder_head_size_min_x = 40mm
extruder_head_size_min_y = 55mm
extruder_head_size_max_x = 75mm
extruder_head_size_max_y = 25mm
extruder_head_size_height = 40mm
But these might be measured in the wrong way/completely inaccurate. Any ideas?

nickthetait, 2015-05-05:
I've tested your change @kakaroto with these settings on the Mini and just assuming that they behave the same for the TAZ too. Now your commit has helped, but there are a number of lingering concerns.

I've found the below values to work well in some (but not all) cases:
extruder_head_size_min_x = 40mm
extruder_head_size_max_x = 75mm
extruder_head_size_min_y = 25mm
extruder_head_size_max_y = 55mm
extruder_head_size_height = 17mm

1. Due to the shape and design of LulzBot toolheads, parts need to be printed in a particular order. Ones at the front of the bed first then moving gradually to the back. The way CuraEngine picks the ordering seems to be inconsistent (or at the very least I don't understand it yet). With just 3 objects it works great but when as more are added the toolhead crashes into already printed models pretty regularly. Complex shapes that span most of the Y axis would not be possible with the current implementation.

2. When someone changes toolhead (to a FlexyDualy for example) I have no idea what would happen.

3. Automatic part re-positioning. I don't know if this is new or should be in its own issue. If you have two objects and you re-position part B so that it is too close to part A, then part A automatically gets moved out of the way. Problem is that it goes the wrong direction half of the time and moves both objects. Pushing part B towards X+ or Y+ is when it misbehaves.

If a objects cannot be made with this style Cura used to silently roll back to using print all-at-once. There were a number of cases where users did not understand when this feature gets disabled.  Recent upstream changes (https://github.com/daid/Cura/commit/2c6b662b1da241fab190f81ce6520bbc70e908f0) added a warning message at the bottom of the object viewer window which helps.

nickthetait, 2015-05-05:
Here's a little picture I drew while taking measurements: 
[image]

kakaroto, 2015-05-21:
This seems to work fine for me with the latest lulzbot-devel branch and the settings you provided above. If you still have issues please document the exact stl used and positioning of each object and how/where it had a collision.
Thanks.
(Leaving the issue open for now until we re-enable the option in the code)

kakaroto, 2015-05-22:
I've pushed the fix and set the values in the config wizard so it would just work for the Mini.
I'll close this issue for now, if there are more issues, we will reopen it and fix it again.

closed 2015-05-22 by kakaroto