alephobjects/Cura #108: Quickprint setting for ABS turns on the fan

kakaroto, 2015-05-21:
I just noticed the fan is turned on (40% to 60% speed) when printing in ABS with the quickprint settings, this was the cause for many of my smaller prints to warp and detach from the bed during printing. I've copied setting to expert mode and disabled the fan to get it to print properly.

nickthetait, 2015-05-21:
I've always heard "dont use the cooling fan for ABS." [Current print profiles](https://github.com/alephobjects/print_profiles/blob/master/cura/Mini/ABS_fast_mini.ini#L47) use these same numbers. So I'm guessing this is desired behavior. Could you shed any light on this @brentmac ?

brentmac, 2015-05-22:
In general that's the case for printers with j-heads of budas that print ABS at their max temperature of 230. We found that running the ABS hotter with a small amount of fan gives the best compromise between strength and print quality. 

It's definitely not ideal for small parts, but I think it's worth it for the better overhangs and higher dimensional accuracy. Maybe we could add the cooling fan on/off option to quickprint to give users that option without going into full settings?

nickthetait, 2015-05-22:
Thanks Brent. Picking particular settings for a specific material is for advanced users, they already have access via full settings. Let's keep simple settings as simple as possible.

closed 2015-05-22 by nickthetait