alephobjects/Cura #109: UI bug in the view menu when Layers is selected

reprapperMC, 2015-05-29:
When Layers is selected in the view menu a small layers graphic is appearing at the bottom of the layer scroll bar at the bottom

[image]

kakaroto, 2015-05-29:
That's not a bug. It's a feature introduced in 15.0. When you click on that
button it will toggle the single layer view. Basically click on it, and the
scroll bar allows you to view a specific layer only vs.  all layers below
it.

reprapperMC, 2015-05-29:
Ok, makes sense. We've already talked about this, but for documentation, we need to replace the graphic as it is confusing and too small.

reprapperMC, 2015-05-29:
See commit 3350ad7352f597b3f6c7c4e17738d5808daed0d6 for the addition of the layer toggle graphic. We'll have to see how this looks as it is very small. It may be best to make the button larger than what it is currently sized down to in Cura.

kakaroto, 2015-05-29:
New graphics was added and resized, it's not less confusion. Closing the issue.

closed 2015-05-29 by kakaroto