alephobjects/Cura #112: Clear Platform Doesn't

gwlinn, 2015-05-30:


gwlinn, 2015-05-30:
Several times, I used "Clear Platform" in order to build a different object.  The platform appeared to close and I loaded the new object.  However, when I began printing, I saw that Cura was printing the previous object.  So, I now routinely close and reopen Cura to make sure the new object will be printed.

kakaroto, 2015-06-01:
If you had the Print controller open, you would need to close it and reopen it. Once the print controller is open, it will print whatever was on the bed at that time, this allows you to keep manipulating your print bed without affecting your current print.

nickthetait, 2015-06-23:
@gwlinn If you run into this problem all you have to do is close the print window and then click "control" again. See #114 for why this is being closed.

closed 2015-06-23 by nickthetait