alephobjects/Cura #113: Infill still printing before perimeters even with new setting turned on

reprapperMC, 2015-05-31:
Building from Cura commit 7b0201d87a6ec0cb5f7f760e729f402b8ab69c33 and CuraEngine 8e244e7fec37c89fe3d3c7dc013ef5bffcf6aa0a the new Perimeters before infill selection check setting is available in advanced settings.

However, in certain instances generated gcode is still printing infill before perimeter after the initial solid infill layers. This occurs when using quick print settings AND when using the full settings with the new Perimeters before infill setting turned on.

This behavior generally occurs in parts with thin walls as far as I can tell. However, printing infill before perimeter creates artifacts in the perimeter. This behavior provides less attractive prints so I'm not sure the reason why this behavior was added.

See the link below for gcode generated with our last 14.09 build and then from the build of the most recent commits.
https://gist.github.com/reprapperMC/f4431991b44e306d3629

nickthetait, 2015-06-01:
The fix is in the [legacy branch](https://github.com/alephobjects/CuraEngine/tree/legacy) ONLY at the moment. I'll merge that into master, build, and test this.

nickthetait, 2015-06-01:
Fixed!

closed 2015-06-01 by nickthetait