alephobjects/Cura #114: Warm when print is modified

kakaroto, 2015-06-01:
Following up on issue #112, I think it would be a good idea to have a warning to the user whenever he presses the Print button on the print controller, IF the print bed or settings were modified.
So if you open the print controller, then change something and then press the print button, a dialog would say "Your print has been modified in Cura, do you want to reload the current platform and settings" (or something like that, in a more user-friendly message) to avoid confusion to the users.

nickthetait, 2015-06-23:
Well, I've come to the conclusion that this is not worth fixing since we will be moving to a whole [new UI system](https://github.com/Ultimaker/Cura) soon.

closed 2015-06-23 by nickthetait

nickthetait, 2015-07-09:
This will be taken care of by #138