alephobjects/Cura #118: When print manually cancelled, heatbed and hotend remain hot

ahancey, 2015-06-07:
If, during a print, I have to manually hit 'cancel', the printbed and hotend do not cool down.  Maybe this is by design, but on the Mini, there is no way to start the print back up in the middle of a print.  It seems like everything should cool down and move back to the 'home' position.   Or, maybe there should be 3 options, 'pause', 'restart' and 'cancel'?

kakaroto, 2015-06-08:
The next release will have a Pause/Resume button, but I think it's a really good idea to have the temperatures set to 0 after a cancel.
I'm not sure about homing automatically though, but I think it might be a good idea to do it. Will wait for input from @reprapperMC or @nickthetait on that.

nickthetait, 2015-06-08:
Absolutely, when "cancel" is selected the nozzle(s) and bed should cool down. I do not see a need for moving the toolhead in this situation.

nickthetait, 2015-06-19:
Is this an appropriate place to make the nozzle/bed cooldown? https://github.com/alephobjects/Cura/blob/LulzBot-stable/Cura/util/printerConnection/serialConnection.py#L99

kakaroto, 2015-06-22:
I think it's the right place, yes. just call self.sendCommand to send the M104 and M140 commands and that should do the trick.

closed 2015-06-22 by nickthetait

ahancey, 2015-06-26:
Just curious, how do I get an updated version with this fix?

nickthetait, 2015-06-26:
As of right this second, you would have to [build an installation package](https://github.com/alephobjects/Cura#packaging) yourself. Plenty could go wrong and many other things are probably broken.

So I'd recommend waiting until our next release comes out. That will be happening soon™. I'll try to remember to ping you here once it is ready.

commit b4e0007d (2015-06-22): Cooldown toolhead and bed whenever print is canceled (including closing print window). Fixes #118