alephobjects/Cura #119: Windows installer needs to be compatible with silent install

orias, 2015-06-10:
The main application is compatible, but the driver install process is not. When used through SCCM Cura installs but fails during the driver install process. If this can be fixed it will make Cura installation through several different new computer imaging tools easier and allow for increased Edu/corporate adoption.

From helpdesk ticket:
 I’m trying to deploy the Cura Lulzbot Edition out via SCCM, and to do so, the installer needs to run silently. It looks like the Nullsoft installer
supports the “silent” flag, but that fails when the installer spawns the Arduino driver installer (ie when the silent installer tries to launch another installer, it’s no longer silent). We already have the Arduino drivers on all of our machines, so it’s an
unnecessary step. I’ve tried removing this step from the installer and re-packaging, but I keep getting errors on errors and I just don’t have the knowledge to keep debugging them.

So, that leads me to my question – do you all have a version of your installer that **does not** force the Arduino driver installation wizard (and/or runs silently from start to finish)? I figure the answer is probably no – but again,
in the off-chance that you do (or can easily provide this), it’ll save me hours and hours of time.

orias, 2015-06-10:
https://www.it.iastate.edu/services/sccm

CJS7070, 2015-06-10:
I'm the person that made the request -- here is a bit more information, and please do let me know if you need any more details.

The reason that the install fails with SCCM is that SCCM requires a "silent installer" to run -- that is, the installer has no prompts, and just installs the software (generally to a directory that's provided as a flag when running the installer). It looks like the installer for Cura uses Nullsoft, which has a /S flag to run silently -- which is awesome, that functionality is already built into the installer, so that doesn't need to be changed.

However, there's a step in the installer to install the Arduino drivers. I looked at the file that builds the installer (scripts/win32/installer.nsi) and it looks like all this step does is determine the computer architecture, and call another installer (an EXE file). The problem is, this second installer doesn't run silently -- so even if you run the Cura installer silently, it will run without any popups until it calls the Arduino driver installation. Since that install isn't running silently, the whole process ends up failing. 

So, my request is that the installer be changed in one of two ways:
1. Modify the installation such that the Arduino drivers are also installed silently (which I haven't looked into), or even just integrate that step in with the main installer (if you are able/permitted to do that).
2. Remove the step that calls the Arduino driver installation; our environment already has the Arduino drivers installed so we don't really need to run those again.

I tried option 2 myself, but ran into errors on errors on errors when I tried to run the package.sh script and eventually gave up (since I'm an IT pro, not a developer). 

Please let me know if you need any other info on my part -- and thank you for looking into this!

kakaroto, 2015-06-12:
I haven't tested this, but looking at the installer script, it looks like the driver install simply doesn't happen if the install is silent :

```
Section "Install Arduino Drivers"
  ; Set output path to the driver directory.
  SetOutPath "$INSTDIR\drivers\"
  File /r "drivers\"
 
  ${If} ${RunningX64}
    IfSilent +2
      ExecWait '"$INSTDIR\drivers\dpinst64.exe" /lm'
  ${Else}
    IfSilent +2
      ExecWait '"$INSTDIR\drivers\dpinst32.exe" /lm'
  ${EndIf}
SectionEnd
```

As you can see, it says "IfSilent +2" to basically skip the driver installation if silent is enabled. This was added by daid in commit id 67c19b1f33dab7175d161e7bd157b12cead81c6e from October 28th.
In short, I think your issue is probably because you used an old version, but the issue is 'already fixed'.

On the other hand, it might be better in the case of silent install to still install the drivers but silently. As you can see, dpinst32/64 is used to install the drivers, and it also has a silent flag /s or /q : https://msdn.microsoft.com/en-us/library/windows/hardware/ff544775(v=vs.85).aspx 
also see https://msdn.microsoft.com/en-us/library/windows/hardware/ff550803(v=vs.85).aspx#setting_the_quietinstall_flag

Leaving this open until it's decided if it's considered fixed or if we want to make it install the drivers silently too.

CJS7070, 2015-06-12:
@kakaroto thanks for looking into this. Either way is fine for me, but I would think that making the driver install silent would be the best way going forward.

Is it possible to get a packaged binary installer for the version that supports the silent install? Every time I've tried to run the package script, I've run into problems.

nickthetait, 2015-06-24:
@CJS7070 the latest stuff can always be found [here](https://www.lulzbot.com/cura). The "15.02.1-1.03" version that is available right now is pretty fresh, only a week old. If possible could you do a little test?

CJS7070, 2015-06-29:
Thanks @nickthetait, I didn't realize that the posted version had been updated. I ran a couple of local tests and it looks like everything worked as it should.

Thanks again, all, for your help!

nickthetait, 2015-06-29:
Good to hear. Take care!

closed 2015-06-29 by nickthetait