alephobjects/Cura #120: Cura build should not rely on Marlin

Jnesselr, 2015-06-11:
The current Lulzbot-Devel package.sh relies on building Marlin in several different configurations. I noticed this because the way the script handles finding the Arduino installation is incorrect (That last else on line 151 should be an else if so that the next if statement reports if Arduino isn't installed).

But even from a standpoint of having Arduino installed (and then being concerned about which one from which group and specifying the version manually) I still think it's a bad idea in general. I know it seems like a good idea to include the functionality of sending a pre-built image, but how many people actually use it?

Even if a few people do use it, I think it would be better to host them on some server so we can manage updates and have the client always grab the newest one when need be, even if they haven't installed the latest Cura.

If nothing else, the path check bug should just be corrected.

Jnesselr, 2015-06-11:
Also, even with Arduino installed, building Marlin on the mac fails because AVR_TOOLS_PATH isn't defined and it should be. On the mac, it's under ARDUINO_INSTALL_DIR/hardware/tools/avr/bin.

If we can confirm that for the other operating systems (C:/Arduino-1.0.3 and /usr/share/arduino) then I'll push up my changes that sets the path to the avr tools directory if avr-gcc can't be found in the path.

Jnesselr, 2015-06-11:
I checked with all of the installs, and it seems that the avr tools will always been in that subdirectory path. I still think we should remove the Marlin compiling stuff from the package.sh script and make detection of different Arduino versions more robust.

reprapperMC, 2015-06-11:
We should see how this is handled in the new QT version of Cura. If they haven't found a better way to do it there, we could work on a fix and see if they will accept a pull request.

nickthetait, 2015-06-12:
Hey @Jnesselr, I agree that this is something that needs fixing. Right this second I need to get a release out of LulzBot-devel. Couldn't build anything on Debian (locally or build server) without reverting your commit. Do not have time to spare on this issue today. Hope you understand, this is nothing against the work that you have put in on this one.

Jnesselr, 2015-06-12:
That's fine, but could you update the issue with what debian spat out so I can fix it?

nickthetait, 2015-06-12:
I really did not want my first interaction with you to be reverting your work, that would've been pretty rude! This is a pretty low priority at the moment, but here's what I got:

./package.sh debian_amd64
Cloning https://github.com/Ultimaker/Marlin.git into _UltimakerMarlin
Removing Marlin/_UltimakerMarlin_115200/
Removing Marlin/_UltimakerMarlin_250000/
Removing Marlin/_UltimakerMarlin_Dual_115200/
Removing Marlin/_UltimakerMarlin_Dual_250000/
Removing Marlin/_UltimakerMarlin_HBK_115200/
Removing Marlin/_UltimakerMarlin_HBK_250000/
Removing Marlin/_UltimakerMarlin_HBK_Dual_115200/
Removing Marlin/_UltimakerMarlin_HBK_Dual_250000/
Removing Marlin/_UltimakerMarlin_Plus_115200/
Removing Marlin/_UltimakerMarlin_Plus_250000/
Removing Marlin/_UltimakerMarlin_Plus_Dual_115200/
Removing Marlin/_UltimakerMarlin_Plus_Dual_250000/
HEAD is now at 19f52b4 Merge branch 'Marlin_v1' into Marlin_UM_Original_Plus
Current branch Marlin_UM_Original_Plus is up to date.
/home/aleph/Software/AlephObjects/Cura
Switched to branch 'Marlin_v1'
Your branch is up-to-date with 'origin/Marlin_v1'.
Current branch Marlin_v1 is up to date.
  CC    /usr/share/arduino/hardware/arduino/cores/arduino/wiring.c
  CC    /usr/share/arduino/hardware/arduino/cores/arduino/wiring_analog.c
  CC    /usr/share/arduino/hardware/arduino/cores/arduino/wiring_digital.c
  CC    /usr/share/arduino/hardware/arduino/cores/arduino/wiring_pulse.c
  CC    /usr/share/arduino/hardware/arduino/cores/arduino/wiring_shift.c
  CC    /usr/share/arduino/hardware/arduino/cores/arduino/WInterrupts.c
  CXX   /usr/share/arduino/hardware/arduino/cores/arduino/WMath.cpp
  CXX   /usr/share/arduino/hardware/arduino/cores/arduino/WString.cpp
  CXX   /usr/share/arduino/hardware/arduino/cores/arduino/Print.cpp
  CXX   Marlin_main.cpp
  CXX   MarlinSerial.cpp
  CXX   Sd2Card.cpp
  CXX   SdBaseFile.cpp
  CXX   SdFatUtil.cpp
  CXX   SdFile.cpp
  CXX   SdVolume.cpp
  CXX   motion_control.cpp
  CXX   planner.cpp
  CXX   stepper.cpp
  CXX   temperature.cpp
  CXX   cardreader.cpp
  CXX   ConfigurationStore.cpp
  CXX   watchdog.cpp
  CXX   /usr/share/arduino/libraries/SPI/SPI.cpp
  CXX   Servo.cpp
  CXX   /usr/share/arduino/hardware/arduino/cores/arduino/Tone.cpp
  CXX   ultralcd.cpp
  CXX   digipot_mcp4451.cpp
  CXX   vector_3.cpp
  CXX   qr_solve.cpp
  CXX   /usr/share/arduino/libraries/LiquidCrystal/LiquidCrystal.cpp
  CXX   /usr/share/arduino/hardware/arduino/cores/arduino/main.cpp
  CXX   _UltimakerMarlin_250000/Marlin.elf
  COPY  _UltimakerMarlin_250000/Marlin.hex
make: /usr/lib/ccache/avr-objcopy: Command not found
Makefile:439: recipe for target '_UltimakerMarlin_250000/Marlin.hex' failed
make: *** [_UltimakerMarlin_250000/Marlin.hex] Error 127

Jnesselr, 2015-06-12:
Oh, okay, I think I see the issue. Can you run 'which avr-objcopy' for me?

And I'm good, I just needed to run the commit before to build on Mac.

nickthetait, 2015-06-15:
`/usr/bin/avr-objcopy` for both my development machine and the build server.