alephobjects/Cura #124: Resume ignores Z-height changes

nickthetait, 2015-06-17:
Printer: Mini (untested on Taz)

Steps to reproduce:
* Using this [branch](https://github.com/alephobjects/Cura/tree/pause-and-one-at-a-time)
* Start a print
* Pause during printing
* Move toolhead up or down any amount
* Click resume

Expected:
Toolhead moves back to the position it was in when pause was selected, then continues printing.

Actual:
Toolhead moves back to correct x & y position but any changes in Z height are ignored. It is possible that this will result in crashing into the bed. While testing, this happened to me damaging the PEI :cry: before I hit the power switch and would have hit the glass too.

I've been told this may be an issue with firmware, but have not investigated this possibility yet.

This is the reason behind [removing pause from LulzBot-Devel](https://github.com/alephobjects/Cura/commit/333a72b508288a2eaa9e1eda36d9193b5553c603).

nickthetait, 2015-06-17:
There are a number of related problems that might need their own issues:
* In some cases, restoring position moves much faster or slower than when printing normally
* After a pause, the rest of a layer will print faster or slower than normal 
* X-axis movements can be off by about 1mm when restoring position
* Hitting an X/Y endstop completely ruins positioning
* Some amount of retraction is needed immediately after pause

nickthetait, 2015-06-19:
Could you take a look into this @kakaroto ?  I've not been able to figure out much since I discovered this a week ago.

What is broken: firmware, software, both? How much effort will it take to fix? Is there a partial solution that could be easily added? One idea was to disable Z axis move buttons while paused, but this is not ideal...

kakaroto, 2015-06-24:
This is an interesting issue because from my understanding, when you pause the print, you're simply not supposed to move the print head... That's what the pauseAtZ plugin is for, it moves the print head for you and restores it to the right position, but if you do it manually, it basically becomes 'unexpected behavior'.
There is no actual reason for which the resume button would revert any manual changes you did. The only reason why it works for X/Y is because the start gcode has the G90 'use absolute positioning' command in it. So when you resume, the next move command will say to move to position X and Y using its absolute positions, and that causes it to revert to wherever it was supposed to be.. however, unless you pause at the exact moment before a layer change, then the G1/G0 commands won't specify which Z position to go to, which means the Z doesn't change, that's what I believe is the issue you are seeing.
To fix it properly, we'd have to track down the X/Y/Z position at the moment we did the pause, then send custom G1 command to put the toolhead at the right position where it was, before actually resuming the print.
While this might work for simple use cases, it won't work very well if the firmware suddenly loses its bearing and doesn't know anymore where the X0, Y0 and Z0 positions are, which can easily happen if you move past the limits of the print bed (for example, it's 200 mm wide, and you move left until you reach 250 mm, the internal X position in the firmware may be 250, even though it stopped moving past 200 mm, I think that's what happened with the park position of (190,190) in the pauseAtZ plugin which caused a shift on the Mini), or if you hit an endstop, then the X/Y/Z positions might be reset, or maybe they won't be, I'm not exactly sure. And even worse, if someone moves the print head manually (with his hands), instead of using the UI arrows, then the firmware has no way of tracking down where the toolhead is positioned in order to put it back to its original position. Either way, these would require doing a home command before resuming, which, on the Mini would also mean doing a ABL probing sequence, which would probably knock off any half printed part..
So, basically, I think it should just be documented that you're not supposed to move the print head when you pause a print.. you'd pause it to check something, maybe to take a picture, but not to move the print head around as that could mess with your print.

nickthetait, 2015-07-02:
It is seriously confusing (to me and future users) when a button is disabled right now. @kakaroto, would you be able to create Enable and Disable methods for each of these:
* X/Y Axis
* Z Axis
* Extrude/Retract
* Temperature
* Everything (which just calls all of the above)
Disable would grey out (with a partially opaque overlay?) the button area in addition to disabling individual button's functionality.
Enable would restore normal colors and make buttons functional.

nickthetait, 2015-07-09:
This will be taken care of by #138

closed 2015-07-09 by nickthetait