alephobjects/Cura #125: Changing flow % setting in Cura does not affect flow

familiarCatch, 2015-06-20:
14.09-1.18.7

Adjusting the flow rate in Cura has no affect of the actual flow rate.

It definitely could be me, let me know what I can do to help.

nickthetait, 2015-06-22:
Are you adjusting the flow rate via the LCD panel on a Taz?

familiarCatch, 2015-06-23:
Adjusting the flow on the LCD works correctly, adjusting flow through Cura does not.

kakaroto, 2015-06-23:
Cura doesn't have a "flow rate" option similar to the one on the LCD. The LCD flow rate option basically slows down or speeds up the printing, but the "Flow (%) " option in Cura under the Filament section is used to compensate for extrusion settings and if you save the gcode from 100% flow to a 95% flow, and diff the two files, you'll see that the flow option is taken into consideration.
The flow % option in Cura is basically the same as the "Extrusion multiplier" in Slic3r, and it's there to compensate for over or under extrusion by fine tuning how much filament is extruded without messing with the filament diameter or the E-steps in the firmware.

nickthetait, 2015-06-23:
So this is a labeling issue. Right now it is displayed as "Flow %". What would better match the actual usage?

kakaroto, 2015-06-23:
I don't think it's a labeling issue, it's listed under Filament, just below diameter, it indicates the filament flow %.. even on slic3r, the extrusion multiplier is as a percentage. Either way, the tooltip explains it pretty well. I think the real naming issue is with the "flow rate" from LCD which confused me a lot at first because it's actually just a "speed modifier" and I couldn't understand why it was called "flow rate".

familiarCatch, 2015-06-23:
Under Filament in Cura, the Flow % option does nothing for me. If it is set to 0, or 100, it prints the same.

nickthetait, 2015-06-23:
I can verify that changing that number does have an affect.
* 0% gives no material flow except for a little dribbling
* 100% is the normal good looking print
* 200% is ugly, deformed, and inconsistent 

What printer are you using? Could you describe your exact steps to reproduce the problem? Please include stl and profile that you are using.

nickthetait, 2015-06-30:
Without any more details I can't provide any more help.

closed 2015-06-30 by nickthetait