alephobjects/Cura #129: Over and Undersized Features:  Function of the positioning code or the material?

Wayne91, 2015-06-25:
TAZ5 working with Cura.

All features are material heavy.  ID's are undersized and OD's oversized by the somewhere near the nozzle diameter (0.35mm).  Is this material expansion during fusion, or is this a function of CURA drawing the extrusion "trace" via the centerline coordinates of the nozzle?   Could someone validate that features "shells" are circumscribed during the extrusion trace and that the program takes the nozzle diameter into account?

The solutions/corrections in CAD will vary depending upon the answer-- material dependent solely, or combined with of g-code trace positioning.

Thank you.

nickthetait, 2015-06-26:
That is a long known problem in the positioning code. The [slicing engine](https://github.com/Ultimaker/CuraEngine/tree/legacy) has a somewhat simplistic view of how plastic gets placed.

One community member has proposed a [fix](https://github.com/Ultimaker/CuraEngine/pull/94), but it has not been accepted. Best place to get a more detailed explanation would be to ask [here](https://github.com/Ultimaker/CuraEngine/issues/29)

closed 2015-06-26 by nickthetait

nickthetait, 2015-07-09:
Did that answer your question?