alephobjects/Cura #130: Bed Temperature does not update during initial heatup

nickthetait, 2015-07-02:
Steps to reproduce: start a new print while nozzle and bed are cool
Expected: Nozzle and Bed temps update continually
Actual: Only nozzle is updated in graph. The status text in the Print window's title does update both nozzle/bed.

Highlighted in green...
[image]

kakaroto, 2015-07-09:
I just investigated this and it's not a bug in Cura. It looks like we simply don't receive the data from the firmware. This only happens on the M109 command which is "wait for the extruder to reach this temperature". If you look at the Marlin code (gcode_M109() in Marlin_main.cpp), you'll notice that the format is to send every second the Target extruder, the temperature of the target extruder and the resiliency time, but nothing about the bed temperature. That command is also blocking which means that if we cancel or try to do anything, then it won't work unless we turn off the power from the printer, so we have no way of requesting the bed temperature in a different command for example. The only way to fix this would be to modify Marlin to add the bed temperature output in that command's loop.
Should we do it or close this bug ?

Jnesselr, 2015-07-09:
Close it. Possibly open up a bug report, but with any luck the work we're
doing on the reprap standards will alleviate this. BotQueue suffers from
the same issue with Marlin.

nickthetait, 2015-07-09:
Good to know, thanks for the research guys! Could you file the report with Marlin @kakaroto?

closed 2015-07-09 by nickthetait