alephobjects/Cura #132: Unwanted Pause command inserted when printing

Makuna, 2015-07-02:
Please let me know information you need and how to retrieve it to help diagnose this issue, as it is causes a dangerous situation of pausing the print with the extruder on and left in its last position.

Cura LulzBot Edition, version 14.09-1.18

Printing a large print to Taz4 over USB.  At about an hour 1 of 31, I noticed the lack of sound coming from the work room, when I investigated, the printer was displaying a "Waiting for User Input" message on the Taz4.  
Both the heated bed and extruder were still being heated, but it paused at its last position (burning a hole in the print).  The Computer running the print had shown no UI changes (still connected, still printing, but with lots of "Error: No LIne Number with checksum" in the message window, which I see all the time).  
Clicking the control wheel on the Taz4 continued the print and it now shows "Resuming print".

nickthetait, 2015-07-06:
Would it be possible to provide the gcode that this issue happened with?

Makuna, 2015-07-06:
When I print directly from Cura, does it cache it somewhere?  Or do I need to export and hope it does the same thing?

kakaroto, 2015-07-06:
You can export the GCode from the File->Save GCode menu. It will be the same gcode as the one used in the print interface, but let's hope it's the same one as when you had that issue (not sure if you changed anything, especially the possibility of using the pauseAtZ plugin)

nickthetait, 2015-07-06:
Also, were you using the "quick print" or "full settings" mode?

Makuna, 2015-07-06:
Full Settings, Lulzbot Cura Fast ABS profile, 2.1 shell thickness, Brim adhesion, otherwise everything else was default.
While "pause at height" and "Teak at Z 3.2" are available, neither are in the enabled list.
the gcode file is 43Mb in size, 14mb compressed, how can I attach it here?  I can't use the trick of appending PNG as that is too large.

nickthetait, 2015-07-06:
Unfortunately github does not let you attach files other than pictures. If you have a Dropbox account it is pretty easy to [share files](https://www.dropbox.com/help/167). Otherwise upload it to http://fileconvoy.com/ then post the special link that it provides.

nickthetait, 2015-07-06:
Another option would be to post it to our [forum](forum.lulzbot.com).

Makuna, 2015-07-07:
The forum isn't working correctly.  I have an account and I just I tried to upload as zip and gcode, in both cases after hitting the upload the page refreshes to an empty post with no attachments.

Here is the link to fileconvoy upload
http://www.fileconvoy.com/dfl.php?id=g0ae5ccdddf38fcb69996881544d84f8673bc4ea7e

nickthetait, 2015-07-07:
I used the "load profile from gcode" feature in Cura to take a look at the settings used. Nothing seems to be out of order there. Then searching through the gcode, I can't find any pause commands. 

My only guess as to what went wrong is that the "pause at height" plugin was unintentionally/accidentally enabled. If this problem happens again please immediately (before closing Cura or doing anything else) export the gcode and let us know about it. Closing this issue for now. :unamused:

Thanks for your patience with the file uploading shennaigans. The forum had a restriction based on file type which I was unaware of. This should be fixed now.

closed 2015-07-07 by nickthetait

Makuna, 2015-07-07:
@nickthetait I get a lot of checksum errors in the print output, is this normal?  Could there be a communications problem that a command got mangled into a pause?

I am positive that the plug ins were not enabled.
I will keep my eyes open for the issue and capture the good if it repeats.

nickthetait, 2015-07-07:
Yes, those errors are normal and we get lots of questions from users about them.

Thanks for your cooperation.