alephobjects/Cura #133: Configuration Wizard enters endless loop

hroncok, 2015-07-06:
On Fedora, the configuration wizard enters endless loop. [Original Fedora bug report](https://bugzilla.redhat.com/show_bug.cgi?id=1231624).

**Steps to reproduce:**

1. install Fedora >= 21
2. install cura-lulzbot package (`yum install cura-lulzbot`)
3. be sure you don't have `~/.cura-lulzbot` directory
4. start Cura Lulzbot (with `cura-lulzbot` or from apps menu)
5. click anywhere in the configuration wizard

**Actual results:**

A second configuration wizard pops up. Continuing to click spawns infinite configuration wizards. There is no path to continue.

**Expected results:**

Able to select machine type and proceed with wizard.

**Notes:**

 * I thought this was a packaging error, but when I unpack the tagged sources form Github, and run `python -c 'import Cura.cura as cura; cura.main()'`, the same error happens.
 * Does not happen with Ultimaker (daid's) Cura.
 * Happens with 14.12 and 15.02.1-1.03 and with anything in between.

nickthetait, 2015-07-06:
Since you run Fedora, could you take a look at this one @kakaroto?

Thanks for the great bug report @hroncok!

hroncok, 2015-07-06:
Fedora >= 22 might actually be needed, as I have some non standard (i.e. newer) related packages on my Fedora 21 (mostly wxPython 3 with GTK 3).

The bug was reported against Fedora 22.

kakaroto, 2015-07-06:
Thanks @hroncok, I'm updating to FC 22 right now to test it and see how it can be resolved. Update is taking a looong time though!

kakaroto, 2015-07-07:
After a long and difficult process in trying to update my fedora from 20 to 22. I can reproduce the issue. It seems to happen on the latest development branch too and seems to be caused by an update to GTK which exposed an existing bug caused by a change in the splash screen code (Ultimaker's Cura doesn't have the issue, but also has a completely grey splashscreen).
Anyways, it's fixed now, so thanks for reporting it!

closed 2015-07-07 by kakaroto

hroncok, 2015-07-07:
Thanks for the fix. Sorry for the bad update experience.

kakaroto, 2015-07-08:
Np, no need to be sorry, it wasn't your fault :)
The update experience was mostly caused by only having a / partition that was almost full (a 16GB mSata SSD on a laptop) and having to go from Fedora 20 to 21 then 22, and having done the FedUp process 6 months ago but without rebooting since then and not knowing about the 'fedup --clean' command :)