alephobjects/Cura #138: Upgrade Print window

nickthetait, 2015-07-09:
It is currently a janky plugin with a number of issues. Needs reworking to become a first class feature.

nickthetait, 2015-07-09:
One idea I have is: do we have the console hidden by default or only appear for developers (running from source)?

nickthetait, 2015-07-09:
I sketched up a design. Here is our current look just for reference:
[image]

When you first open Cura this is what you would see:
[image]

Load up a model, pick a printer/profile, make modifications then click the "print" icon which reveals the print panel on the right side.
[image]

Make some tweaks, preheat/swap filaments, do whatever you need to prep the printer, then click the "start print" button. This reveals a progress bar at the bottom.
[image]

Here is a little detail sketch for each of the submodule controls. These are pretty similar to what we have now.
[image]

[image]

[image]

Dual extrusion stuff would disappear when only a single extruder is configured.

kakaroto, 2015-07-10:
Humm.. I'd have to disagree with the design, I don't think having the print window integrated into the Cura main window is a good idea because it will use up too much of the real estate of the window. I think it should remain as a separate window, but have it more tightly linked to the main application as a real window object, rather than a plugin script that calls a small set of callback functions. This would allow us to have things such as hide/show the console for example, or clicking on the temperature graph would open it full screen, and of course, hide/grey out the movement buttons when in printing mode, etc..

reprapperMC, 2015-07-17:
This issue has now been duplicated at https://code.alephobjects.com/T27. Please move any future discussion there as issues will shortly no longer be tracked on this repo.