alephobjects/Cura #139: Minimum layer time is not respected

kakaroto, 2015-07-10:
The minimum layer time in the advanced panel is not respected when printing small parts. I've set it to 30 seconds, and as you can see in this video : https://dl.dropboxusercontent.com/u/22642664/cura/VID_20150710_164931.mp4 it does 3 layers in about 6 seconds.
This was using the latest Lulzbot-Devel branch and latest legacy branch of CuraEngine.

ahancey, 2015-07-11:
I have experienced the same issues when printing smaller parts, the minimum time is not being used and the part prints too quickly.  If you would like a sample uploaded, I would be glad to do that.

kakaroto, 2015-07-14:
Thanks, that's fine. It's easy to generate a test part for this in openscad : 
```cylinder(r=2, h=20);```
It looks like the setting is not actually ignored (changing the value does create a different gcode output), but I want to first finish issue #138 before I look into this one more deeply.

reprapperMC, 2015-07-17:
This issue has now been duplicated at https://code.alephobjects.com/T26. Please move any future discussion there as issues will shortly no longer be tracked on this repo.