alephobjects/Cura #142: Cannot deselect object

3dprintdaddy, 2015-07-16:
User reported issue to Lulzbot Support:

Windows 8
Cura 15.02.1-1.03

User cannot deselect current object.
When loading multiple objects, can never select the second model, or even deselect the first model.

Jnesselr, 2015-07-16:
I cannot seem to replicate this issue on 8.1. Is it strictly Windows 8, or
does that include 8.1?

nickthetait, 2015-07-16:
Same here on Win 8.1. Sounds like it is specific to 8

Jnesselr, 2015-07-16:
Confirmed as Windows 8 specific. If I delete the first object, the same
effect now happens to the other object. Ie, I can select it but not
deselect it.

When loading the second object, the first object shows the lighter green as
if it's selected, but not the white outline. The second object is a dark
green. Clicking anywhere selected the first object.

reprapperMC, 2015-07-17:
This issue has now been duplicated at https://code.alephobjects.com/T24. Please move any future discussion there as issues will shortly no longer be tracked on this repo.